amigavision/AmigaVision #17: UFO: Enemy Unknown (OCS) glitches

meauxdal, 2021-03-07:
Little bug report. 

UFO: Enemy Unknown works - it launches and the gameplay seems fine, but the music plays far too slow and the HUD in Battlescape seems to be tinted green. 

I ran the OCS version directly from Workbench, using the /shared folder to transfer the WHDLoad files directly. It ran the music fine from here, but did do some weird flashing in certain transitions. No green tint on the HUD here either.

Using MegaAGS-20201116 with its included default config

Directly from the MegaAGS:
https://www.youtube.com/watch?v=5cDBZvskNF4

From Workbench on the same MegaAGS image, using my own WHDLoad files:
https://www.youtube.com/watch?v=MReQBxmDyy8

Optiroc, 2021-03-11:
Can you check the status of this issue, @meauxdal ?

meauxdal, 2021-03-11:
Just had a quick test - issue seems to be fully resolved on MegaAGS-20210309 - fresh install with default configuration files. Music is now playing at the correct speed, there is no green tint in the Battlescape HUD. No flashing either, although that only happened with my own WHDLoad files. Thank you @Optiroc !

closed 2021-03-11 by meauxdal