amigavision/AmigaVision #20: Some games have too aggressive cropping settings

movisman, 2021-03-10:
Hi!

As discussed, I tried some random games and the following have crop settings which are way too aggressive (too much is cut off the top and bottom of the screen):

Cannon Fodder
Cannon Fodder 2
Cannon Fodder 2 Alien Levels
Cannon Fodder New Campaign
Cannon Fodder Plus
Cannon Soccer

Skidmarks
Super Skidmarks
Super Skidmarks Farmyard
Super Skidmarks Plus

I'm sure there will be others as there are hundreds of games - as I find more I can report them if you like. My display is running regular 1080p.

A few others like Mortal Kombat 1/2, Flashback also crop a little information from the top/bottom, but not too much. I understand some cropping is to be expected, if so these ones are probably fine - unless you want to be super conservative in which case these might need to be dialled back too. I think they are probably ok, but ones like Cannon Fodder and Skidmarks are missing way too on screen information now.

Any info you need, just ask. Happy to help.

Thanks!

limi, 2021-03-11:
Yeah, so here’s where things get a little subjective — and the lack of me publishing info on the general philosophy of 5× scaling/cropping is probably not helping.

If I were to _guess_, you are reacting to a few screens like the title screen and the level start screen. In general, if any title or intermission screen gets cropped in a less than optimal way — _but_ there’s a way to make the in-game action fit reasonably well — we _will_ take that trade-off. 

A few pixels can be sacrificed during gameplay, of course depending on the nature of the game. If the game becomes harder to play with a crop, or we hide e.g. critical HUD info or scores, we don’t do it.

Allow me to demonstrate with this game as an example, it’s an interesting edge case — and I’m open to arguments as to why we chose the wrong scale factor. 

(Ignore the aspect ratio in the screenshots, it’ll look like this without the scalers etc. The 5×PAL is _wider_ than 4×PAL, but won’t show as such here. I know this is confusing, but I wanted to have the sharp pixels to show exactly where the cut-off is. To get the real effect of the 5×PAL, you have to take a photo of the TV. Think of it as the “shorter” screenshot making better use of the 1080p surface, so it is preferred if we can get away with it. Also, don’t forget that the 5×PAL scaling only applies to HDMI output and has no effect on e.g. a 4:3 CRT.)

Here’s the title screen in 5×PAL/cropped and 4×PAL, respectively:

[image]
[image]

5× looks bad, right? Missing part of the logo? But this screen is in no way critical to the gameplay. Unfortunate, but not critical.

OK, next is the selection screen. This one would be bad to hide information on, and there’s information all the way to the bottom edge if you have a large enough squad. 5×PAL and 4×PAL, respectively:

[image]
[image]

As we can see, no pixels are lost on this screen, and it works fine with 5×PAL. (…and even if we didn’t have enough space here, cutting off a little bit of the top of the screen would have been OK here, luckily we don’t have to do that.)

Next up, another bad situation, the level title screen — 5×PAL and 4×PAL, respectively:

[image]
[image]

You lose the title of the level here. We saw this as a reasonable trade-off, as the mission briefing on the following screen is visible. (Sorry, too many screenshots already, so I will skip that one)  

OK, what about in-game? 5×PAL/cropped and 4×PAL, respectively:

[image]
[image]

As you can see, we sacrifice parts of the stretched globe along the bottom — but it’s still identifiable as a map/globe when seen on a large TV in 1080p. We also sacrificed a few pixels at the bottom, which could be critical in e.g. a platformer like Flashback, where you could have platforms that were just 1-2 pixels on the viewport, and have a bad effect on difficulty level or gameplay in general.

However, e.g. driving games and top-down games like this don’t rely on seeing every single pixel for the game to work fine. This is when we accept cropping. And if you try the game in 4× vs. 5× on an actual TV instead of these screenshots, I think you will agree that losing a few pixels in-game to the 5× scale is a trade-off worth making in this case.

(You can of course make the game load permanently in 4×PAL by removing the `PAL5` argument in the `.run` files for a given game — or just override it temporarily with Force PAL, which will default to the 4× scale.)

limi, 2021-03-11:
And here’s how it looks in-game on a 1080p display:

**4×PAL**
[image]

**5×PAL**
[image]

limi, 2021-03-11:
For Skidmarks, I could go either way. I’m probably 51% on the side of filling up more of the screen than having a smaller viewport but seeing more — specifically for this kind of top-down small track racing game.

### 4×PAL on 1080p surface:
[image]

### 5×PAL on 1080p surface:
[image]

Optiroc, 2021-03-11:
Great write-up @limi ! I think overall you've picked the right trade-offs, so big up for that 👍👌👍

Cannon Fodder is a tough nut... I think I'm on the 5x side for that one. I think the missing level names is unfortunate because there's a lot of humor and tone added by those. The few lines cut off in the in-game UI is largely inconsequential, but it does make it look a bit broken. Again, a tough one.

movisman, 2021-03-11:
@limi this is a great write up, thanks for covering all of this. The screenshots are really helpful. You are right, I did make some of my initial conclusions based on the title and mission screen for Cannon Fodder, and also for Skidmarks (i did also notice the game was zoomed in a bit). I just thought something must be up or some behaviour not completely intended, hence why I brought it up. But now I understand how it works. I guess quite a few other games will show this effect too in similar areas.

It is a very interesting discussion though and extremely subjective. It is a really tough one to call with some of these, because the bigger image is really nice, but I do also feel that having such a large amount cropped off the title screen and on things like 'mission' screens could also raise a lot of questions from people who use the image. They may not understand the reasons why it happens in some places and not others (like myself - although your explanation above has helped no end). It could potentially cause a bit of confusion or people might think something is wrong.

As far as a preference goes, I need to think about it some more. In cases like Cannon Fodder / Skidmarks, this is where i'm a bit torn. The main game looks nicer with the bigger image, but the title screen and missing information on the mission screen is quite jarring for me personally. Perhaps it just looks a little 'too' strange as quite a lot is cut off - however I do appreciate it doesn't affect the actual game, which looks better in 5x. Although you can just about see the world on the bottom, it does look a tiny bit odd like @Optiroc said, but not too bad. And Skidmarks - it doesn't look like it's got huge borders in 4x mode actually (based on screens above?), so I can see the appeal of having a smaller viewport but seeing more of the track, especially as the borders aren't as big as some other games. But also, having a larger image (but seeing less of the track) is also acceptable and appealing too as it totally fills the screen. I can see why you could go either way on this. 

With games which are just shaving a handful off top and bottom, I think this is always fine and a sensible compromise. It makes total sense and usually nothing is missed at all. And you get a much larger image in return.

For cases like Cannon Fodder, it's definitely a situation where you can't please everyone and trade-offs have to be made! :)

I *think* I possibly actually steer slightly more to the side of preferring to see everything and just dealing with more borders as a compromise, when there is heavy cropping involved to get to 5x. I'm not 100% convinced either way yet though. There is just a jarring feeling when something pops up on the screen and a good inch is missing from the top and bottom, like in Cannon Fodder. But on the flipside I do feel comfortable with cropping on the games which only have a slight crop like MK, Flashback etc.

Although I haven't tested, I wonder what will happen in games running 5x (that suffer heavily cropped title screens like Cannon Fodder), that for example, have a 'shop' or something that breaks up the action in the middle of the game, which may potentially use more of the screen. I guess it's possible this could be heavily cropped as well, potentially missing valuable information? Or game endings etc with pictures, text etc could possibly crop too much off? Hope i'm explaining that properly.

Thanks for the note about NTSC sizing by the way, that is useful to know that actually 5x will always display nicely with minimal borders in this mode, and it is only some PAL games which need a 4x/5x decision.

I have a question, I know in the options we can 'force PAL' for that session, but is there a way to have it save as default if the user wants? I think the answer is probably no, but just curious.

The other option if people wish, is if they don't want any cropping at all in favour of large borders, they could just not upload the customised minimig_vadjust.dat file to the shared folder so default sizing is used.

Thanks both - interesting discussion. Maybe in a few days I will have a firmer opinion, once i've had a play on a few of the Euro PAL games on the image.

Optiroc, 2021-03-12:
Not saving the forced video mode option is by design. It's too easy to forget that you've set it, and it causes problems with a lot of titles.

If you really want to, you can copy the `whdlvmode` file from `ENV:` to `ENVARC:` after you've set a forced video mode and it will stick.

movisman, 2021-03-12:
I wondered if it might be by design. You are right though, it would be easy to forget you've set it, and to be honest you've already done the hard work by classifying which games run best in PAL, PAL60, or NTSC. So auto should be fine in the majority of cases, and any which have issues we can report so that they are fixed. I personally planned to leave it as auto anyway as it's better.

limi, 2021-03-15:
Skidmarks has been reverted to 4× PAL, we kept Cannon Fodder at 5×.

closed 2021-03-15 by limi