amigavision/AmigaVision #22: UFO: Enemy Unknown request

meauxdal, 2021-03-15:
The OCS version of this game is quite excellent, but is cut down quite a bit from the DOS original - see https://en.wikipedia.org/wiki/X-COM:_Enemy_Unknown_(1994_video_game)#Ports_and_re-releases

There are subsequent releases for the Amiga, however - the AGA and CD32 versions. The latest MegaAGS image currently uses only the OCS version.

As far as I understand, here are the differences:
OCS: Reduced colors from the DOS version. "Thinking" progress bar before/during enemy turns. Animations are turned off by default. Battlescape has (creepy, excellent) background music. Everything is pretty slow.
AGA: Full colors from the DOS version. Some visual improvements to the HUD. "Thinking" bar has been removed. Animations are turned on by default now. All-new music, most of which is good, but the Geoscape music is arguably worse. The Battlescape does NOT have background music for some reason. Game seems to run a bit better overall.
CD32: As far as I can tell, this is roughly the AGA version with an additional introductory cutscene (which is pretty great). Saving may be handled differently - IIRC there was no prompt to enter a filename and it instead handled that automatically.

I suggest adding one or both of the later versions of the game. The higher color support alone makes them a lot easier on the eyes, and the CD32 introduction scene is impressive and has great music. I think this particular game is important enough to have all 3 versions present in MegaAGS, but I am probably (definitely) biased! I also may be missing differences or other reasons why the OCS version might be preferable as this game's representative.

Thank you!

limi, 2021-03-15:
Yes, having played both the OCS and the AGA version back in the day, the AGA version is meaningfully improved, and we should probably switch to that being the default.

Optiroc, 2021-03-15:
Let’s 👍

closed 2021-03-15 by Optiroc