amigavision/AmigaVision #24: Audio issues with Robocop 3 shootout sections

movisman, 2021-03-19:
Hi guys, just adding this as an issue here as per a previous conversation with @Optiroc.

Easiest way to replicate:

If you load it up and select arcade mode, and then select one of the shootout sections (I think F2 on the selection screen is one). When the game starts, hold down the mouse button to shoot and you'll see how messed up the sound is. It plays a load of random game samples followed by some horrid white noise/glitched sample sounds, which does eventually stop (thankfully). It should just play the sample of Robocop's pistol.

I think this is another one of those 68020 CPU speed / timing things.

It doesn't happen in WinUAE using the preset expanded A1200 configuration, so it isn't a WHDload issue.

If I boot from floppies in A500+/ECS mode, it is fine of course.

I've stumbled across a few games now which have messed up or glitchy sound in 68020 mode on MiSTer - as i've mentioned previously games like Hostages and Shufflepuck Cafe are slightly glitchy, same in WinUAE, but ticking cycle-exact fixes it.

Not sure if you can really do anything about the issue with this game given the fact MegaAGS has to boot in 68020/AGA mode. @Optiroc suggested it could be a regression with MiSTer regarding sound in 68020 mode.

I thought i'd flag it here though in case you wanted to add a note to the DB.

There are no bugs filed on the WHDload Mantis bugtracker for this issue, but as I say I think it may be MiSTer specific, as it doesn't happen when you mount MegaAGS in WinUAE.

Perhaps this issue would be best placed in the Minimig issues for MiSTer - let me know what you think.

Cheers!

Optiroc, 2021-03-19:
I’ll add the sound issues to the note or issues field in the DB, so they can be followed up in future core releases. 

I used to include an automated list for titles with issues noted in the launcher. I should probably reintroduce it!

movisman, 2021-03-19:
No worries. Hopefully the sound in general when using 68020 can be improved in the future.

It might be worth me collecting some of this into a single issue and raising it on the Minimig Github for MiSTer. Although it's likely that this kind of behaviour is 'expected' because of the CPU running faster than a real A1200. Workaround of course is to use A500/A500+ type config + floppies, but then we don't have the benefit of using MegaAGS, which is what I prefer by a mile.

Optiroc, 2021-03-25:
Added issue note in fd924a1b1894bcf4eb54a2c3bb476ee608182d21. Let's hope someone can fix this in the core at some point. Maybe something to bring up with Ole who implemented the PWM audio recently?

closed 2021-03-25 by Optiroc

movisman, 2021-03-25:
Yes, that's not a bad idea. Will have to put together some wording to post on the forum or something, or maybe I'll drop him a PM. It would be great if these audio issues (including the minor glitches on those other games I've mentioned before) can be resolved on the core itself.