amigavision/AmigaVision #25: Full Contact will not continue between 2nd bonus round and 3rd opponent

movisman, 2021-03-19:
Hi,

Another one to flag on your DB. This one is actually listed an issue on the Mantis bugtracker for WHDload with A1200 configuration.
http://mantis.whdload.de/view.php?id=4741

It has been assigned since July 2020 but no amendments since.

Basically, after you complete the 2nd bonus round, as it tries to load the 3rd opponent the screen goes black, the music continues, but it never gets past this point.

It means you can never get past the 2nd stage of the game.

I'm not sure if this would still happen when using the WHDload version of the game on a A600HD type config - I don't have the environment to test this. But as MegaAGS needs 68020/AGA to run, this is probably a moot point anyway.

Cheers!

Optiroc, 2021-03-25:
Added issue note in fd924a1b1894bcf4eb54a2c3bb476ee608182d21

closed 2021-03-25 by Optiroc

movisman, 2021-03-25:
No worries! Hope this gets fixed sometime. I will append a note to the existing bug on the Mantis bugtracker entry, just to confirm I can replicate it.

movisman, 2021-04-05:
Hi,

I added a note to the existing bug the other week regarding this freezing after stage 2, and apparently it is now fixed!
http://www.whdload.de/games/FullContact.html

See the version notes for 1.7. There are some other changes as well.

Dunno if this will be picked up automatically for future updates, but thought i'd let you know anyway just in case there is any manual intervention on your part.

Cheers!

Optiroc, 2021-04-05:
Yep, it will (kind of) automatically be rolled in with future updates. Reason it's not fully automatic is mostly that I like perusing the changelogs to get to know stuff like this. ;)

movisman, 2021-04-05:
Got it, no probs at all! Thanks as always.