amigavision/AmigaVision #26: Superfrog occasional freeze in one spot

movisman, 2021-03-19:
Hi,

I have raised this via the bug report form on the WHDLoad install page for Superfrog, as it is an issue with the WHDLoad package (i think). I can replicate on MiSTer, WinUAE, etc.

Load stage 2.1 - code 392822
Hold down jump and aim for the first candle above where you spawn. Sometimes it will freeze quickly, other times it will take minutes to replicate so it's a bit random. You can come at it in different directions, or jump straight up, i've had it freeze in that spot in a variety of situations.

It is low impact as it doesn't happen often, but it did catch me out twice before I decided to try and replicate it and report.

Requires a full reboot once it freezes. Temporarily uploaded a very amateur video via WinUAE here:
https://www.youtube.com/watch?v=6INkB0H63hY

It does the same on MiSTer.

I understand that the disk version is used in MegaAGS. There are two separate entries for Superfrog - the main game and the intro. I think this must be because the disk version had the intro on a separate disc, but this wasn't required to be loaded if you just wanted to play the game. Hence why they are listed separately.

I am not sure if the same issue above occurs in the CD32 version or not. I have not tested this. But the CD32 version is apparently supported by the WHDLoad installer. Maybe it is worth considering using the CD32 version of Superfrog, as it has a skippable intro and there would only be a need for one entry in the games list (not Superfrog and Superfrog - Intro). Not sure if you have any thoughts on that. There are no enhancements on the CD32 release though. It might not be worth messing about with tbh.

Again, just logging this here in case you wanted to add it to the issues DB. It is such an unusual bug to find though, and won't happen regularly, so it might not be worth trying to come up with a few words of text to describe the issue.

I did log it via the WHDLoad site though - maybe someone will look into it.

Cheers

Optiroc, 2021-03-19:
Sounds like the CD32 version is the better pick in this case. Thanks!

movisman, 2021-03-19:
No worries! No idea if it will have that same random glitch on the CD32 version using WHDLoad, but I suppose the benefit of using that version either way in MegaAGS is only having one entry in the games list, instead of two. If you do change it I can always test it again and see what happens.

I should add that I couldn't replicate this on a regular 'cracked' floppy release.

Cheers

Optiroc, 2021-03-19:
I assume that the same issue will be present on the CD32 install too, but let’s find out.

Going from two Superfrogs to one is one step closer to the ideal amount of Superfrog, so it’s a win either way. 😈

movisman, 2021-03-19:
Ha!! I quite agree!
My girlfriend loves it however - she isn't really a gamer but did have an Amiga growing up and she remembers both Superfrog and 'Gloom' vividly (no idea why), so as soon as she spotted Superfrog I couldn't get her off it for a while!

Optiroc, 2021-03-25:
Changed default version in fd924a1b1894bcf4eb54a2c3bb476ee608182d21.

closed 2021-03-25 by Optiroc

movisman, 2021-03-25:
Thanks!

Went back and forth on the issue I posted on the Mantis bugtracker. In the end I was able to replicate it on the CD32 iso as well as floppies. And the debug info for WinUAE apparently indicates it's a game bug.

Still, probably worth making CD32 the default Superfrog release as you can have a single entry for it then.

Thanks!