amigavision/AmigaVision #27: Music in Formula One Grand Prix runs too fast - PAL/NTSC?

movisman, 2021-03-19:
Hi,

Just booted up an old classic, Formula One Grand Prix by Geoff Crammond.

However, I noticed straight away that the intro music plays back too fast - the game is set to run in NTSC but if you force PAL, it runs at the right speed.

Not sure if this is deliberate or not, but I thought you'd want it pointing out in case it should be corrected.

If you do change it to PAL, perhaps the cropping setting for the game could be checked and verified also. It looks good in NTSC as naturally it fits at 5x with no cropping. However the HUD does run right to the bottom of the screen, so if it was changed to PAL you wouldn't want that info to be chopped off. However it is a bit of a border fest in default PAL mode.

Maybe this is why it was set to NTSC in the first place! :)

Anyways, thought i'd raise it for you.

Thanks a lot.

Optiroc, 2021-04-07:
Gotcha! I'll fix it to run in 50Hz but with 5X scaling.

movisman, 2021-04-07:
No worries! So long as if it's in 5x mode, the HUD doesn't get cut off at the bottom. I believe the HUD runs right to the base of the screen so if there is cropping involved, it might need positional tweaks or to be left at 4x?

Optiroc, 2021-04-07:
Don't worry, the game runs in a 320x200px viewport 😘

movisman, 2021-04-08:
Ah right, in that case it should be perfect then! :) I am not sure if the menus run in a larger viewport and could be cropped as a result, perhaps that might need to be verified.

Optiroc, 2021-07-05:
Fixed in 50b6eb010a7db219f2c244e01d55530897f99704

closed 2021-07-05 by Optiroc

movisman, 2021-07-08:
Great, thank you very much as always! :thumbsup: