amigavision/AmigaVision #47: Make shared folder mounting + disk buffers conditional or failsafe

limi, 2022-03-06:
To allow the script to be used to create images for CD32, original Amiga and MiST users, we should see if there’s a way to make these fail gracefully or be conditional.

Optiroc, 2022-03-07:
This is pretty niche and easy enough to edit for any user advanced enough to want to do this.

But if anyone knows how to detect that we're running on a CD32 this would be easy enough, like by checking a specific version (or existance) of a certain library.

Optiroc, 2022-03-07:
I guess for the "addbuffers DF0:" issue, we can just check if DF0: exists before running addbuffers...

Optiroc, 2022-03-07:
GitHubs advanced AI regarded the issue as done based on my commit message. Fair enough!

retrofun, 2022-03-07:
@Optiroc: Thanks for looking into this but this does not work. The problem at least on MiST is that the system freezes either immediately or after a few seconds when mounting MR0: (mount MR0: in S:MountMisterFS)

Solution that works on MiST:

Use isMiSTer from MinimigUtils.adf, directory MiSTorMiSTer
https://github.com/minimig-dev/MinimigAGA-MiST-TC64/blob/master/amiga_sw/MinimigUtils.adf

Copy MiSTorMiSTer/isMiSTer to C:
Change S:Startup-Sequence from

```
;Mount shared folder
execute S:MountMisterFS
```

to

```
;Mount shared folder
isMiSTer
if warn
  execute S:MountMisterFS
endif
```

This would have to be tested if it works for the different platforms.

Optiroc, 2022-03-08:
@retrofun Thanks! It's plain obvious that I'm not avoiding to mount MR0:, when taking another look at this commit. :)

I'll look at what the "isMiSTer" command does and rework.

commit ffe11427 (2022-03-07): Update AmigaDOS scripts, adding checks for MiSTer volume and DF0 (fixes #47?)
closed 2022-03-10 by Optiroc