amigavision/AmigaVision #68: Adding games issue

theenemy-thd, 2022-11-20:
On older versions I was able to add the files to dh1:whdload/letter and add files needed (iff,run and txt) in dh0:ags all/letter and when booted back up it would show the new added game

This downstairs happen now which is a shame as I have purchased games (devils temple, turbo sprint) which I enjoyed from running via megaAGS

Optiroc, 2022-11-20:
Yes, there is a new directory cache feature that throws a wrench into a manually edited setup like yours. 

The solution is to either remove the `.dir` file in the directory you want to edit. Alternatively you can edit the dir file, it’s just a text file.

theenemy-thd, 2022-11-20:
I tried editing the .dir last night and it just exited and  threw up a load of characters ... I will try deleting like you mentioned .. thanks

closed 2022-11-20 by theenemy-thd