amigavision/AmigaVision #69: Add Hired Guns (Demo)

keungk, 2022-11-30:
as its a completely different to the release version.

limi, 2023-01-26:
I believe the WHDLoad installer already supports that, as it says in the docs:

Note:
> In tooltypes, you can use:
> - for demo only: [CUSTOM1](http://whdload.de/docs/en/opt.html#Custom)=0 Campaign Demo (1 player)
>                         =1 Action demo (2 players)
>                         =2 Attract mode

Let me know if this is not the thing you are talking about, or if you need some help in how to set tooltypes on the Amiga :)

keungk, 2023-04-17:
i think its one installer and two separate games. Whdownload site has them as separate downloads

limi, 2023-04-18:
OK, I’ll take a look.

Optiroc, 2023-05-07:
Added

closed 2023-05-07 by Optiroc