amigavision/AmigaVision #76: Some excellent PD freeware games could be added

rsn8887, 2023-04-07:
For example, the fantastic freeware games 
Babeanoid 2
Air Taxi
Cybersphere Plus
could be added, WHD

They are WHDyfied in the Aladin Generic WHDLoad collection, more info here:
https://eab.abime.net/showthread.php?p=1124581#post1124581

Download for those three freeware games:
https://files.coolatoms.org/amigapd.zip

Optiroc, 2023-04-07:
Thanks for the suggestions!

Cybersphere Plus and Age of Rock looks worth playing, I'll probably add those. 👍

Might go for Air Taxi too, since it's rare and pretty cool with >2 player games.

As for the Aladin collection. I've added a select few of those back when the MegaAGS collection started, and adding any more will be on a case-by-base basis.

rsn8887, 2023-04-08:
I just learned that there's a prequel to Cybersphere Plus, just called Cybersphere which is also cool:
https://www.mediafire.com/file/zm902qm8tt5xz0q/Cybersphere-WHDL.lha/file

Might be good to add both Cybersphere and its sequel Cybersphere Plus. Both are cool, just with different levels.

closed 2023-04-16 by Optiroc