amigavision/AmigaVision #78: Zoom settings from Amigavision also affect other configurations

rsn8887, 2023-04-08:
There are now .mgl files used and separate crop configs "shared/Miminig_vadjust.dat" and "shared/Amiga_vadjust.dat" which should to be good to isolate AmigaVision configs from other configs, and it seems to work for everything apart the zoom/crop.

After running Mega AGS, my own Amiga configurations are still affected by the last crop config that was active in AmigaVision.

In order to reset the crop for my own configurations, I have to manually delete the two "shared/Miminig_vadjust.dat" and "shared/Amiga_vadjust.dat" files every time after I run AmigaVision.

Maybe I am doing something wrong? I have tested various sequences of running .mgl and minimig core directly, but nothing seems to help with this.

rsn8887, 2023-04-08:
Since the isolation between AmigaVision configs and personal configs works pretty well already, it might be good to not ship AmigaVision with a "minimig.cfg". It already comes with Amiga.cfg and the corresponding .mgl, which is enough. 

I have several friends who destroyed their own personal startup config by copying over the AmigaVision minimig.cfg over their own.

Optiroc, 2023-04-08:
The clashing of the Amiga `setname` that is used both when loading the "raw" minimig.rbf and the Amiga.mgl we now use for MegaAGS was a huge woe that easily delayed this release for at least two months... :D Indeed, it would be best to use a non-clashing setname.

limi, 2023-04-08:
As mentioned in #81, the next release will not include anything related to `minimig*`on the config front, so should address this. 

And, as with that issue, this is not new, and older versions of MegaAGS would also do this, but we can avoid it now that we have the new MGL setup, agreed.

closed 2023-04-08 by limi