amigavision/AmigaVision #80: Some European games set to NTSC or PAL60

rsn8887, 2023-04-08:
If a game was developed by American devs like Epyx, it makes sense to set it to NTSC. Most games are set to PAL/NTSC correctly now, but some might still be questionable, I think.

For example, Lemmings 2 is set to NTSC, but should be PAL IMO (British devs).

The Gobliiins series is set to PAL60, but should be PAL IMO (Coktel Vision, French devs).

Psygnosis games like Amnios and Anarchy should all be PAL, and so on.

limi, 2023-04-13:
> If a game was developed by American devs like Epyx, it makes sense to set it to NTSC. Most games are set to PAL/NTSC correctly now, but some might still be questionable, I think.

Agreed, there are still games that should probably be running in PAL. But there are also an argument to be made for running games in NTSC/PAL60 when the game can handle it: It can be played on US consumer CRTs (which is also the reason why the launcher defaults to NTSC). So if the game doesn’t run too fast, and the music is correct, this may be the right tradeoff. Of course, the other side of that argument is that people running NTSC should just select Force NTSC and deal with whatever happens. You can also argue that the aspect ratio will be wrong in NTSC, but some of these games actually look more correct in NTSC, as lots of Amiga games didn’t really pay close attention to e.g. how to render a circle in either aspect ratio.

> The problem with setting PAL games to PAL60 or NTSC is that they run too fast, including the music (which sometimes is the best part of the games).

Yes, if it runs too fast (e.g. not 3D games), or the music is at the wrong speed, 100% it should be set to PAL by default.

> The situation is different from old consoles, where the definitive version is often 60 Hz, because the consoles were biggest in Japan or US. Since the Amiga was biggest in European PAL countries, the definitive version of games is most often 50 Hz.

Indeed.

> For example, **Lemmings 2** is set to NTSC, but should be PAL IMO (British devs).

This is an interesting one, because they have taken extra care to make sure the music runs at the correct speed, and as far as I can tell from a cursory inspection, it also doesn’t run at the wrong speed. Note that I haven’t done a super detailed comparison, but trying to play it in both modes, it just seemed like everything was working the way it should, the music was correct, and the only difference was that it was smoother in 60hz.

I kept this at 60hz for now.

> The **Gobliiins** series is set to PAL60, but should be PAL IMO (Coktel Vision, French devs).

I believe this game was made for PC, Atari ST and Amiga. Both the PC and Atari versions are 320×200 by default, and were probably designed for this, I don’t think Coktel was Amiga-first. The characters definitely look too fat for me in PAL, but I’d be open to any arguments to the contrary.

> **Psygnosis** games like **Amnios** and **Anarchy** should all be PAL, I think, because they were British devs.

I believe these are set to NTSC because they are shooters, and historically we have tried to push the shooter category to 60hz because of the smoother gameplay, which matters a lot in these fast games. For these two in particular though, they aren’t fast (nor good ;), so I changed them back to PAL.
  
> German games like **Amberstar** should all be PAL for sure. Nobody in Germany set their Amigas to 60 Hz to play games, be it developers or gamers. You just turned on the Amiga and put in the game floppy, and it automatically ran at the supposed speed of 50 Hz.

Another case where the game was developed for Atari ST, PC and Amiga. **BUT**, the music is clearly too fast, and the orbs at the top of the screen look bad in NTSC resolution, and the game is not a fast game that would benefit from NTSC. So I think this should be PAL. Thalion was also deep into the Amiga game development scene, unlike Coktel Vision, which were generally on a lot of different platforms that were using 320×200 instead of the Amiga PAL resolution of 320×256. 

Feel free to report other games! Just know that we usually have some reasons — sometimes historical, sometimes bad, but sometimes based on other factors where it does make more sense to have them in NTSC, especially for late multiplatform titles like Gobliiins. 

Always open for discussion on what the best trade-off is!

closed 2023-04-13 by limi

limi, 2023-04-13:
> I believe these are set to NTSC because they are shooters, and historically we have tried to push the shooter category to 60hz because of the smoother gameplay, which matters a lot in these fast [...]

Plot twist: Amnios actually had an NTSC release. Anyway, still switched them both to PAL, as I doubt anyone in the US ever played this game. 😅

commit a1c3a3d7 (2023-04-13): These games should be PAL, closes #80