amigavision/AmigaVision #82: Add QR codes that link to game manuals

limi, 2023-04-10:
Since we already have a game → HoL mapping, we could generate QR codes to show in the launcher that would load the relevant game manual on a phone/tablet to use while playing the game.

Example:
http://hol.abime.net/72/manual

We would have to find a way to capture the URL to the PDF, and generate a QR code that can be displayed alongside the game entry.

AGS2 would need some sort of mechanism for showing the QR code in a predefined area of the layout.

peterdutton, 2023-05-15:
This seems like a great idea - have you considered suggesting it at the core MiSTer level? You could imagine an option for 'Manual QR' from the OSD that could work very nicely.

limi, 2023-08-11:
This looks to be the most common Python module to generate QR codes: https://pypi.org/project/qrcode/

HeyItsLollie, 2024-01-26:
I'd love to have this in AmigaVision, it'd be just as useful on PC & Pocket. Already had to download a few manuals just to get past password checks for some games.

BlueMaxima, 2024-03-22:
WHDLoad files usually contain manual, readme, info and guide files in some combination. I'd be surprised if there wasn't a way to be able to read those files from the launcher.

limi, 2024-06-09:
Useful online generator for testing, looks like we’d need at least 25×25px to support 32 characters; 21×21px allows 17 characters, 29×29 supports 53 characters:

https://pewscorner.github.io/barcodes/qrcode_generator.html