amigavision/AmigaVision #87: Demo - State of the Art - Image is cropped

M-Walrus, 2023-04-15:
Part of the image is cropped-out throughout the entire demo.  Tested in 1440p50, both fullscreen and integer

Optiroc, 2023-04-16:
Since I'm a relentless contrarian this is one demo I never liked, and never loaded up on Minimig. I guess I have to now!

My guess is that it's rendering into a fair bit of the overscan area, so we need to adjust a the crop a bit.

M-Walrus, 2023-04-17:
Was just perusing the demos section and watched it my first time there randomly.  My assumption after watching it was that it was doing a 5x scale/crop and should have that disabled.

Optiroc, 2023-04-20:
I had a look and most texts are cut off by design. But the demo does use an extreme amount of overscan, and it would fit better in the 4x crop if adjusted about 8 pixels down (mostly apparent on the dragon pixel art screen).

Optiroc, 2023-05-02:
Fixed

closed 2023-05-02 by Optiroc