amigavision/AmigaVision #88: Spanish Games

Newsdee, 2023-04-15:
"Lorna" in the French touch selection is actually a Spanish game.  There are actually some interesting games from Spain for Amiga which could be their own list. 

**Dusk of Spanish Golden Age**
_Spain was the 2nd largest 8-bit game producer in Europe in 1983 to 1992. These later titles were infamous for being pretty but tough._

_[Publisher: Dinamic Software]_
After the War (1989)
Arctic Moves (2017 port from ST)
Astro Marine Corps (1990)
Freddy Hardest in South Manhattan (1989)
Grand Prix Master (1989)
Hammer Boy (1990)
Mega Phoenix (1991)
Narco Polic (1991)
Navy Moves (1989)
Risky Woods (1992)
Satan (1990)

_[Publisher: Topo Soft]_
Lorna (1990)
Viaje al Centro de la Tierra (1989)

_[Publisher: Opera Soft]_
Angel Nieto Pole 500
Corsarios [1989]
Livingstone II [1990]
Mot [1989]
Sol Negro [1988]

_[Publisher: MA Soft]_
Abu-Simbel Profanation (1998)

_[Publisher: DRO Soft]_
Arte de la Guerra (1990)
Telemark Warrior (1989)

Optiroc, 2023-04-16:
This is brilliant, thanks!

Speaking of Lorna; you’re right of course! I always just took for granted it was French with its total Métal hurlant vibe. But both the comic and game are Spanish, so I guess I may have to cut it, or add a note (because it fits the vibe so well :) )

Newsdee, 2023-04-17:
The Spanish vibe is mostly ports from 8-bit machines, which had a lot of classics but they often did not translate that well into the 16bit era.  Risky Woods and Narco Police (at least) are really nice looking,

Optiroc, 2023-05-11:
I added the list (omitting games not in our database) with a tentative title: 2958d8c3a052dda0c51c65fe1587d9716fb5c7a0

Newsdee, 2023-05-11:
I'm not too worried about the title :)   my original intent was to convey it was the end of an era;  I've been reading the book "8 Quilates" by Jaume Gutierrez (sadly not available in English) which talks about the Spanish software industry of the 80s; the Amiga output was towards the end of it ... and not the best games;  but they definitely had a style to them.

closed 2023-05-12 by Optiroc