amigavision/AmigaVision #94: Extend `title_short` display from 23 to 25 characters

limi, 2023-05-03:
Going through the top lists, it struck me that we are not fully displaying the proper names for a number of the most popular games on Amiga.

Currently, I believe the limit is 23 characters, but just upping that limit to 25 characters instead should take care of most of them, examples:

```
It Came from the Desert
Sensible World of Soccer
Ultimate Soccer Manager
Secret of Monkey Island
Four Crystals of Trazere
World Class Leader Board
Arkanoid: Revenge of Doh
Premier Manager 3 Deluxe
Shadow of the Beast III
```

…versus the current situation:

```
Sensible World of So..
It Came from the Des..
Ultimate Soccer Mana..
Secret of Monkey Isl..
Four Crystals of Tra..
World Class Leader B..
Arkanoid: Revenge of..
Premier Manager 3 De..
Shadow of the Beast ..
```

There does seem to be enough space to do this in the current layout — of course, the `skin-bg.iff` file would have to be updated, and some offsets for where to position the thumbnail images updated.

limi, 2023-05-03:
It’s funny how most Amiga game titles fit into 25 characters (+ the 4 needed for the ranking), wondering if there was a standard `char(25)` entry for game titles among retailers or something.

Optiroc, 2023-05-25:
This needs support in AGS2. The current setting is the maximum supported width.

Optiroc, 2023-08-01:
Done in 1c7004ee253cf6ea67e96822179b530a756321a7

(AGS2 changes in https://github.com/Optiroc/ArcadeGameSelector/commit/f2543ab7e3a001c65833d8e00a51dc248b67b7df)

closed 2023-08-01 by Optiroc