amigavision/AmigaVision #96: Video Settings clarified - vscale_mode

Toryalai1, 2023-05-06:
Hi,
Thank you for this amazing project. I read through the documentation about the video settings and this is what I see:
[Amiga]+
[Amiga500]+
[Amiga500HD]+
[Amiga600HD]
video_mode_ntsc=8
video_mode_pal=9
vsync_adjust=1 ; You can set this to 2 if your display can handle it
custom_aspect_ratio_1=40:27
bootscreen=0

I wonder what the vscale_mode should be set as.

Optiroc, 2023-05-07:
That’s up to personal preference! If you run 1080p all Amiga resolutions will be (vertically) integer scaled regardless of the vscale_mode. At higher resolutions like 1440p I would definitely let the scaler fill the screen vertically.

It might be worth clearing that up in the doc.

thehughhefner, 2023-05-07:
I have LG CX which is a 4K tv and I believe most users have a 4K tv. Could you please include vscale_mode information since the different vscale_modes does different things and you would get different results. Thanks

closed 2023-05-11 by Optiroc