amigavision/AmigaVision #100: Game Addition Requests

hittm4n, 2023-05-15:
4096 (Or another nice version of this game) (Sometimes called 2048)
Act Of War
Bograts (Surprised this was missing)
Boulderdash C64 (No One Inc)
Diamond Caves II (Collection)
Drag And Drop Deluxe (https://electricblacksheep.itch.io/drag-drop)
DropZone
Friday Night Pool
Hibernated 1 + Directors Cut + Addon (https://8bitgames.itch.io/)
Hillsea Lido (Surprised this was missing)
Jelly Othelly
Kastle Kumquat (Looks are deceiving, starts simple and easy, gets tough and very puzzley - I helped Edgar Vigdal make a PC remake of this :) )
Klondike Deluxe + Cardsets
LCD Dreams
Lumberjack + variations
The Lost Treasures of Infocom (I have an HDF collection i compiled which expands this set by 10 more Infocom adventures)
Megaball v4.1 + as many boardsets as can be found (I've compiled boards into my Megaball HDF but the main game is still v4.0)
Motorola Invaders 2 AGA (Metallica and Space Invaders, what could go wrong)
Nemac IV (Surprised this was missing)
Pacman 500
Plat-Man (Brilliant PD Game)
Quest For Galaxia (Available as a single exe on an ADF i made - can supply)
R3 - The Art Of Rocketry + Bonus Levels (My most favourite "Thrust" style game)
Relics Of Deldroneye (And sequels)
Saturday Night Snooker
Scramble 500
Schelobers Quest For A Babe
Sim City 2000 (Surprised this was missing)
Super Foul Egg
Tapper
Tetren (The best Tetris on Amiga)
Timekeepers (Surprised this was missing)
Uridium
Valhalla & The Fortress of Eve (Surprised this was missing)
Willys Weiry Nightmare
Xevious AGA

Optiroc, 2023-05-15:
Some good suggestions here. 

In general, anything not part of the WHDL project has been added piecemeal if someone (me mostly so far) has seen a particular reason to. 

Lost Treasures of Infocom should be in there already. Need to check what happened to it!

SimCity 2000 is deliberately missing because the Amiga port is a disgrace to a great game.

hittm4n, 2023-05-15:
Lost treasures is there, but not my enhanced version :) with 10 extra Infocom adventures added.
Totally understand about Sim City 2000, however it seemed like a bit of an oversight to not include it. Not a game i much like personally but felt it should be included for others to enjoy.
Many of the others are personal favourites. As discussed with Limi on the Mister forums, AmigaVision could be a one-stop shop for many people with a much expanded games section, and especially the demo section and music disk packs. I'm talking with Limi now on Discord and starting to help out where i can. Limi will update you i'm sure...

ghamson, 2023-06-08:
Just got this going on the Analogue Pocket - would love to see these on that version of AmigaVision

- PP Hammer
- Rick Dangerous AGA
- Turbo Sprint (perhaps we need to add a way to prove purchase or something)
- Rotator (perhaps we need to add a way to prove purchase or something)
- Maria Renards Revenge
- Atris (perhaps we need to add a way to prove purchase or something)
- Metal Gear
- Boxx4
- Devils Temple (perhaps we need to add a way to prove purchase or something)
- Bomb Jack Beer Edition

Selfishly I would also love to see a whole section on the great Shareware / Licenseware software that the Amiga scene back in the day.

Very happy to get involved in testing / discussing anything above.

hittm4n, 2023-06-08:
Hi There. All the games below are in the main version at least. Only Atris and Devils Temple look to be missing from the main pack. I guess its a size restriction thing for the Pocket version, but could also be last minute additions to the main pack which missed the Pocket pack. If its size related, then i guess some included games can be dropped and other more popular ones added by request. The pack will never please everyone though unless it has every game going in it (we can but hope). Someone else will have to step in and let you know why games are missing.

P.P. Hammer (OCS)[en]
Rick Dangerous (AGA) (AGA)[en]
Turbo Sprint (AGA)[en]
Rotator (OCS)[en]
Maria Renard's Revenge (AGA)[en]
Metal Gear (OCS)[en]
Boxx4 (OCS)[en]
Bomb Jack Beer Edition (OCS)[en]

I'm totally with you on the shareware/PD stuff though. Many of my own requests list are shareware or PD stuff, theres some great games to be found here, often times rivalling the commercial stuff. I'm hoping to add support for a few titles myself for the next release while screenshotting demos :)

ghamson, 2023-06-08:
In total agreement here.  I enjoy many of these games using a MiSTer, so the post above is just focused on the Pocket primarily.  Either way, the Analogue Pocket version is a massive achievement and I see it entertaining me for many months :)

Optiroc, 2023-06-08:
There’s plenty of space left until the 4GB limit is hit for the Pocket build, so I’ll organize a custom “add list” including the titles mentioned here (as well as other requests I’ve received). 

As for adding games not currently in the database: as I’ve mentioned before it’s relatively easy if they are not using custom track loaders. But it’s a bit of work and I only add things I have personal interest in. But if you are willing to make custom installs, get in touch and I’ll show you the general principles.

AlePax, 2023-06-19:
> Just got this going on the Analogue Pocket - would love to see these on that version of AmigaVision
>
> * PP Hammer
> * Rick Dangerous AGA
> * Turbo Sprint (perhaps we need to add a way to prove purchase [...]

I'd just add

* Stormlord

to this almost perfect list! ;)

dangaeta, 2023-07-13:
I'd like to add:

[Persian Gulf Inferno](https://www.lemonamiga.com/games/details.php?id=829)
[Terry's Big Adventure](https://www.lemonamiga.com/games/details.php?id=1210)
[Bomber Bob](https://www.lemonamiga.com/games/details.php?id=1493)

Thanks :-)

limi, 2023-07-18:
I have applied a strikethrough to the configurations that exist on the Pocket as of 2023.07.17. Please add additional game requests as separate issues, this one should be for tracking the original poster’s requests.

Optiroc, 2023-08-02:
@hittm4n I think we should add a list of quality shareware/freeware to either the curated or focus sections. You seem most knowledgeable on the matter; could you list 20 or so of the best titles? I’ll add what’s currently missing if possible.

hittm4n, 2023-08-09:
Working on a list. Using some EAB lists from way back (for a PD top games list closer to the classic Amiga era), then i will try to put a list together of more modern free games, theres some great stuff. Going to ask in forum for input from everyone for the modern stuff.

hittm4n, 2023-08-09:
In no particular order, heres a top 30. I can continue to refine this if you wish, into a top 20 if you need to keep to that limit, i've tried to make this list classic era Amiga. Working on a modern top 30 too.

rsn8887, 2023-08-14:
Zerberk
Monsters of Terror AGA

are also very good, and 4 player simultaneous! They are by Starfox, the same dev as Minerunner and Poweroids.

commit 0a8606bb (2023-08-23): Add PD lists (closes #100)