amigavision/AmigaVision #106: Add "Napalm" with patch

limi, 2023-06-20:
The game "Napalm" — which we don't have a configuration for at the moment, since it was originally targeted to RTG Amigas and seemed unplayably slow on an AGA Amiga — seems to work well if you patch in the chunky-to-planar pixel support:

https://www.youtube.com/watch?v=yiPkbL4wObI

The patch is here:
http://aminet.net/package/util/boot/Blazewcp

(I don't have any strong attachment to this game, but probably worth taking a quick look)

Optiroc, 2023-07-31:
I'm a bit hesitant to experiment with system wide patches like BlazeWCP. There used to be a couple similar patches included in the startup sequence (FBlit and FText), but they didn't play nicely with some games.

limi, 2023-08-01:
Yes, understood. FWIW, this seems extremely narrowly focused, only changes the execution speed of those three function calls, not the functionality itself. It’s commonly included in other Amiga game setups, but we could also leave it as an exercise for the reader.

(it should of course only be called in this particular game setup, but there’s a theoretical risk that it would affect anything launched after exiting that game)

Optiroc, 2023-08-02:
As far as I can tell BlazeWCP, in contrast to FBlit an FText, doesn't have an option to restore the original OS functions. But we can apply it upon loading specific titles.. There are many other things (demos in particular) that can leave the system in a somewhat weird state after exiting, so that's fair game I think.

There's one good reason to put Napalm in, I think: It's included in the "EAB Top Exclusives" list, where it is currently displayed as "not available". So why not!

Optiroc, 2023-08-02:
Well, this is a CD game that apparently requires the CD to be mounted to boot. If someone has a "no CD" patch we could include it.

Optiroc, 2023-08-02:
I fooled around with Napalm (and Genetic Species which is the other "not available" title) for a bit. Will not touch again.

Optiroc, 2023-08-02:
Also, the platforms we’re targeting are not what these games were made for (ideally 060 with RTG and AHI sound). Closing.

closed 2023-08-02 by Optiroc