amigavision/AmigaVision #108: SWOS 96-97 has no English teams except Arsenal

chanunnaki, 2023-07-17:
I don't know if I'm missing something in my setup, but when I go to pick a club team from England, the only team available in the Premier League is Arsenal and the lower divisions have no teams at all.

I haven't checked the club teams of all the other countries, but almost all countries have their club teams present.

limi, 2023-07-24:
From #109:

> btw since the update (as think is written in another ticket) sometimes when you select a team, it can't find the league file team025.dat etc) and gives you an empty league.

limi, 2024-05-29:
It has been fixed upstream now, it seems — so will be included in the upcoming release.

limi, 2024-06-05:
Fixed in AmigaVision 2024.06.06.

closed 2024-06-05 by limi