amigavision/AmigaVision #109: Swos not loading games Analogue Pocket

Romanista345, 2023-07-24:
Hi,

in the first release on pocket swos 22 could load and save games and self made tactics.. (importing tactics didn't work, due to the famous tactics whd load bug see http://www.whdload.de/games/SWOS.html)

In the second release on pocket in early Jule did break this function
- the swos tactics bug seemed to be solved.. I could even import famous tactics by moving them to the saves.hdf with directory opus
- but in swos 2022 en swos 2023 (which was added in the July release), after a load when you actually started a game, it gave some noice on the top of the prematch screen (which shows the team selections, see attached photo)..
- after a while it gives me the following error exception 'level 3 autovector (copper/vblank/blitter) ($6C) at $41F39FEA (ExpMem $84FEA) occured .. in a standard amiga dialogbox, I could 'make a coredump', which I press, but nothing the I cgot a'recoverable alert' 
- the whd load people wrote in their update that they 'reworked file load/save completely (which fixed load/save tactic bug)' 

Since I modded the saves hdf ( to include those tactics), I just did a clean install of amiga vision (which had a new version last week as you know), this morning, being back home, but it gave the same error

btw since the update (as think is written in another ticket) sometimes when you select a team, it can't find the league file team025.dat etc) and gives you an empty league.

hope you can help.

[image]

limi, 2023-07-24:
The issue with the teams is likely in the WHDLoad container, but we’ll take another look to ensure we aren’t doing anything wrong with the configuration.

The other issues are likely specific to the Pocket version of the core (at least that would be my guess, since we haven’t received any reports about it yet), so I added it to this general issue, subscribe there too to receive updates once @Mazamars312 takes another look at the current games that are known to work on MiSTer, but not on Pocket:

https://github.com/Mazamars312/Analogue-Amiga/issues/19

Romanista345, 2023-07-24:
for my personal use I found this workaround now,

start amiga vision, then put in the floppy version of swos 2023, rest
a d then when loaded put in a save disk 

then saving to adf works

Romanista345, 2023-12-29:
saw that there is a new amiga vision, installed it, but alas the same issue...

limi, 2024-01-01:
Yeah, there haven’t been any updates to SWOS upstream in the WHDLoad containers since last time. Weird that it has had issues all this time.

Romanista345, 2024-01-01:
The weird thong is that the april version didn’t have the problem. I could add taxtics by copying them on my mac fs-uae via directory opus. Brw. There is a new swos 23-24 version released last week. Whd load available via sensiblesoccer.de. Verstuurd vanaf mijn iPhone

limi, 2024-05-29:
It has been fixed upstream now, it seems — so will be included in the upcoming release.

Romanista345, 2024-05-29:
yes i tagged you guys on twitter with thi great news;) thanx

limi, 2024-06-05:
Fixed in AmigaVision 2024.06.06.

closed 2024-06-05 by limi