amigavision/AmigaVision #113: Filter

Manveru-V, 2023-08-04:
Hi,
I'd like to share this with family & friends but would like a way to filter out the R-rated games (like all the Strip Poker games).
Is there an option to remove/hide them, either while AmigaVision is running or in the config?
Or, at least, can the default image be blacked out or just show the name of the game?

Thank you.

Optiroc, 2023-08-09:
The menu system is (literally) a file browser, so the only way to do filtering is when you build the image.

limi, 2023-08-10:
In other words, just delete the entries you want removed from the `AGS:` folders. The launcher does not do any run-time filtering, and it's unlikely to gain that feature in the near future.

closed 2023-08-10 by limi