amigavision/AmigaVision #120: WINGS OF DEATH - "PAL" Game running in "NTSC" mode !!!

Turricaner, 2023-08-31:
WINGS OF DEATH - is crashing on exit.
This is because it is running in "NTSC" mode  and not 
in "PAL" mode like in the description in the Further Info Box.
After the correction everything is fine.

limi, 2023-09-04:
Thanks for that info! Since the game is an Atari ST port, it needs to run in NTSC (what we call “PAL60” in this case), so we don’t want to switch it over to PAL and have the graphics be in the wrong aspect ratio.

Probably worth filing a bug upstream, so I did:
http://mantis.whdload.de/view.php?id=6251

closed 2023-09-04 by limi

Turricaner, 2023-09-05:
Ok, didn't know about PAL60 and the resulting aspect ratio.
By the way ... great work !

Turricaner, 2023-12-10:
OK - But where do you want to release the December update ? (last one doesen´t exists any more on archive.org)

rsn8887, 2024-02-09:
Just because it is an Atari ST mode doesn't mean it should run in PAL60.

This game is a German production and should run in PAL.