amigavision/AmigaVision #127: Some games could be added to Pocket

rsn8887, 2023-10-18:
Games that would be great to have on Pocket:
Marvin's Marvellous Adventure

cosmicvibes, 2023-10-20:
I was wondering about some games myself - what is the criteria for them to be in the Pocket version vs the full version? If there is work to be done on packaging them in some way I might be able to help. Thanks!

limi, 2023-10-22:
> I was wondering about some games myself - what is the criteria for them to be in the Pocket version vs the full version? If there is work to be done on packaging them in some way I might be able to help. Thanks!

It’s simply a matter of deciding that we want them included. The initial Pocket setup was made by including the configurations for the most highly regarded Amiga games and demos, and keep the setup under 4GB. Which means we may have missed some people’s personal favorites — but if you request specific game configurations to be included, it’s easy to add them to the list.

Buyukcaglar, 2023-10-25:
> Games that would be great to have on Pocket: Marvin's Marvellous Adventure Spacetaxi 3 Age of Rock Two-To-One

**Prehistorik** would also be a great addition.

Buyukcaglar, 2023-12-07:
Wow! Thank you very much for taking time and adding those games, especially Prehistorik! ❤️