amigavision/AmigaVision #128: Switchblade II on AmigaVision is cropped on iPad display

keungk, 2023-10-26:
Running Switchblade II from the launcher crops the screen so I can't see the energy or weapons bar, but exiting the launcher and switching to PAL and running it from workbench is fine.

from launcher
[image]

Quitting launcher to WB gives me black bars top and bottom, and switching to PAL  gives borders top and bottom, and left and right.

NTSC
[image]

PAL
[image]

Below are my Mister.ini settings.

`[Amiga
+Amiga1200
+Amiga500
+Amiga500HD
+Amiga600HD]
video_mode_pal=2048,1536,50   ;9
video_mode_ntsc=13
vscale_mode=0
vsync_adjust=1 ; You can set this to 2 if your display can handle it
custom_aspect_ratio_1=40:27
bootscreen=0`

keungk, 2023-11-20:
i added this to my mister.ini and this seems to have helped

video_mode_ntsc=13
video_mode_pal=2048,1536,50 

additonally, i generally force PAL on AmigaVision.

limi, 2023-12-25:
> i added this to my mister.ini and this seems to have helped
> 
> video_mode_ntsc=13 video_mode_pal=2048,1536,50

Yes, you will need to manually add a PAL resolution definition, since the presets only include an NTSC variant.

I also realized I had prioritized cropping for this game wrong, there’s useful info along the bottom of the screen that was lost in 5×PAL. (there is info at the top too, but less useful for gameplay, so it was adjusted)

Note that this crop will not affect the 4:3 iPad screen, so these issues are two separate ones.

closed 2023-12-25 by limi