amigavision/AmigaVision #134: Agony is cropped way too hard

rsn8887, 2023-11-04:
Agony default crop/screenmode and PAL forced speed is wrong. This is a PAL game and should use all the PAL lines, also the speed of the title music is wrong (too slow) when forcing PAL.

Correct image:
[image]

Current AmigaVision behavior (too few lines):
[image]

keungk, 2023-11-20:
can you check Switchblade 2 as well. That is far too cropped for me. try forcing PAL in the amiga vision options

limi, 2023-12-25:
Addressed in https://github.com/amigavision/AmigaVision/commit/dbf4adb86389465087963f228801f6357a1f90de

Note that while we have made PAL the default now, it’s still using 5×PAL Dynamic Crop by default, and we consider it a reasonable tradeoff that some of the title screens are cut off a bit to prioritize 5× scale for gameplay, in line with the other configurations. If you don’t want Dynamic Crop, you can Force PAL in the Options, which will disable the crop/scale.

But music and speed should be consistent with what you expect in both cases now.

closed 2023-12-25 by limi