amigavision/AmigaVision #144: Unpleasant control mapping for racing games.

Atariboy1982, 2023-12-28:
Is there any possibility of mapping this differently on the Analogue Pocket? On the Pocket d-pad up to accelerate is very much a less than ideal setup.

limi, 2024-01-16:
Can you make use of the controller remapping in the core?

Analogue button → Core Settings → Controls

If not, your best bet would be to file requests for the individual games to add support for a second button similar to how a lot of games support up-to-jump as well as a second button:

https://www.whdload.de/search.html

limi, 2024-05-29:
Nothing we can do here. I recommend requesting it for the games you want it for here:

http://mantis.whdload.de/view.php?id=6502

(Mantis issue tracker is a bit of a disaster usability-wise, so let me know if you need assistance)

closed 2024-05-29 by limi