amigavision/AmigaVision #150: [MiSTer] [2023.12.25] [1080p] [Virocop] 5xPAL scaling causes lives and ammo counter to be left off the bottom of the screen.

IainMcNulty, 2024-01-04:
On MiSTer (1080p) - 5xPAL scaling causes lives and ammo counter to be left off the bottom of the screen on Virocop.

Here's what we see on the current version of Mega AGS/AmigaVision using the included scaling settings:
[image]

And here's what the same game looks like with the ammo and lives counters visible:
[image]

limi, 2024-01-08:
I *think* I managed to find a good compromise between 5× crop and showing the energy/ammo meters — but Virocop is one of those games I never got around to playing seriously, so I’d appreciate your feedback.

If you want to test the new setting on your current setup:

Edit `Amiga:AGS2/Run/Game/Virocop (AGA)[en]`

Change the `vadjust` line to:

`ags-vadjust s=5 v=24 h=0`

Let me know if this seems acceptable. It crops a little bit of the score and number of lives along the top, but seems worth it to get 5× scale, since they are still readable.

IainMcNulty, 2024-01-10:
These settings are far better, I would consider this an appropriate fix.

As a heads-up, the number up top isn't number of lives, it is viruses remaining in the level for Virocop to eliminate, the lives counter is in the bottom-left corner of the screen 😉

limi, 2024-01-10:
> These settings are far better, I would consider this an appropriate fix.
> 
> As a heads-up, the number up top isn't number of lives, it is viruses remaining in the level for Virocop to eliminate, the lives counter is in the bottom-left corner of the screen 😉

Ah! As long as it is still reasonably readable — which it should be — I’ll consider this fixed.

closed 2024-01-10 by limi