amigavision/AmigaVision #157: Overkill AGA wrong cropping

rsn8887, 2024-01-20:
Overkill AGA is cropped too much, so that the status bar is not fully visible:
[image]

rsn8887, 2024-01-20:
Never mind I accidentally saved config, which saved previous wrong cropping.

rsn8887, 2024-01-20:
Not even the "Force PAL" option fixes this game cropping.

limi, 2024-04-04:
Yeah, something else is going on here. This game seems to make extreme use of the overscan area? 

I pulled it down 10px, so the HUD is fully visible; hopefully this doesn’t affect the game on other stages, as I am stealing some pixels from the bottom of the screen. This is running at 4×PAL, not even 5×PAL, and still overflows.

closed 2024-04-04 by limi