amigavision/AmigaVision #158: Superhires detail missing in Super Stardust AGA

rsn8887, 2024-01-20:
Super Stardust AGA Score and life display on top left should look super detailed like this (super hires sprites) like here on my own WHDLoad install running normal Minimig core:

[image]

But when loaded in AmigaVision they look like this, no detail:

[image]

You can also see it immediately looking at the arrow in the title menu:

My install, standard minimig:

[image]

In AmigaVision, the arrow looks wrong, missing superhires detail again:

[image]

In my own WHDLoad install (same slave and game version), it looks correct, and floppy version also looks correct. I am using same MiSTer.ini config and same core version in all cases.

Only AmigaVision looks wrong.

Maybe it is a consequence of cropping? I tried Force PAL, but the problem remains.

rsn8887, 2024-01-20:
Found the fix: For AmigaVision, I need to use Pixel Clock: "28 MHz", not "adaptive." Then it works. In my own install, this setting doesn't matter.

limi, 2024-01-20:
Yeah, Super Stardust is one of the few (the only?) games that use Super Hi-Res. Will investigate and see if we can make that the new default without hurting other games.

Appreciate the detailed report!

limi, 2024-04-04:
28MHz is the default in the upcoming release. Thanks!

closed 2024-04-04 by limi