amigavision/AmigaVision #160: Game/Music too fast for NTSC/PAL60 Titles

synrais, 2024-01-27:
When searching online for Puzznic Amiga music, every one I find is the slower music, not fast like the NTSC/60 patched version on AmigaVision.
This is a song that was written and only available for Amiga I am pretty sure.
[](https://www.youtube.com/watch?v=9FnWvm9RWmo&list=PL5OZqGJNojJ3dd9pjdnDl0rcauEfxj0-1&index=34&ab_channel=PlinthYT)

And here is the AmigaVision version
[](https://youtu.be/GdnXMUtD7ug)

Utopia is another one I feel
[](https://www.youtube.com/watch?v=c7RGtHOj8pM&t=874s&ab_channel=JisengSo)

Here is the NTSC SNES version of this song known as Astoria II
[](https://youtu.be/PJcJuOm3XH0?si=TikoQ5r0QmddoY3B&t=635)

AmigaVision Remix
[](https://youtu.be/gZD9v43mrzM)

This is just the first 2 games I checked there are probably many more

Stoppschildgruen, 2024-02-25:
The games are incorrectly set to NTSC instead of PAL. You can change it yourself by switching from MegaAGS to the Workbench via ESC on the keyboard, opening CygnusEd (right-click on the top screen under Applications), opening the corresponding file in `AGS:Run/Game` and changing `NTSC` to `PAL` and `s=0` to `s=5`.

**OLD:**
```
...
ELSE
  ags-vadjust s=0 v=0 h=0 
  whdload >NIL: "Puzznic.Slave" NTSC BUTTONWAIT SplashDelay=$whdlspdly $whdlqtkey
  ags-vadjust
ENDIF
...
```

**NEW:**
```
...
ELSE
  ags-vadjust s=5 v=0 h=0 
  whdload >NIL: "Puzznic.Slave" PAL BUTTONWAIT SplashDelay=$whdlspdly $whdlqtkey
  ags-vadjust
ENDIF
...
```

limi, 2024-03-09:
Utopia is fixed in https://github.com/amigavision/AmigaVision/commit/ecc9499321734cc51c6c9ad64e0c22f18549b4f0

commit 1f421f7f (2024-03-09): Puzznic was erroneously running in NTSC mode. This closes #160
closed 2024-03-09 by limi