amigavision/AmigaVision #161: Enhancement request: allow the ability to delete entries in the frontend.

theshinyknight, 2024-01-29:
Would be great to be able to remove/delete entries from the main frontend.

Currently you can add favorites, but that get out of hands pretty easily (and you can import duplicates, so it is not ideal); while there is no way to remove an entry.

There are many games that you may not play, and those could be removed directly in the interface, as you use the frontend. Either the entry can be deleted or the whole game/package can be deleted maybe, but the latter would be better to also trim down the overall size of the drive.

limi, 2024-01-30:
We will not support file management from the launcher, you can do that using Directory Opus or similar tools.

You can remove a favorite by hitting the TAB key or the tertiary button when you are inside Favorites.

closed 2024-01-30 by limi

theshinyknight, 2024-01-30:
Thanks; are you going to add some instructions about how to remove the entries from the frontend via DOpus or similar? Without knowing what elements compose the entry on the frontend, it is not that simple to delete it. Thank you

limi, 2024-01-30:
The Favorites are in `Saves:WHDSaves`, the non-custom ones are in `AGS:`. Documentation for AGS can be found here: https://github.com/MagerValp/ArcadeGameSelector