amigavision/AmigaVision #164: Chicago 90 5xPAL

Stoppschildgruen, 2024-02-25:
The game Chicago 90 is incorrectly set to 0x and not 4x as stated. But 5x looks even better.

For those who want to correct it on their own: Switch from MegaAGS to the Workbench via ESC on the keyboard, open CygnusEd (right-click on the top screen under Applications), open the corresponding file in `AGS:Run/Game` and change `s=0` to `s=5` and `v=0` to `v=20`.

**OLD:**
```
ELSE
  ags-vadjust s=0 v=0 h=0 
  whdload >NIL: "Chicago90.Slave" PAL BUTTONWAIT SplashDelay=$whdlspdly $whdlqtkey
  ags-vadjust
```

**NEW:**
```
ELSE
  ags-vadjust s=5 v=20 h=0 
  whdload >NIL: "Chicago90.Slave" PAL BUTTONWAIT SplashDelay=$whdlspdly $whdlqtkey
  ags-vadjust
```

synrais, 2024-02-25:
Please document all changes you make, it will help fix these speed/region issues with the modified titles.

limi, 2024-03-09:
0 just means normal PAL— not 0×, by the way :)

commit b46a12cf (2024-03-09): 5×PAL setting for Chicago 90. This closes #164
closed 2024-03-09 by limi