amigavision/AmigaVision #167: Examine list of NTSC titles to figure out which ones should be “Sachs NTSC”

limi, 2024-03-09:
There are quite a few US-produced games that should be using the 320×200 “Sachs NTSC” (tall pixels), now that we have that available to us.

IainMcNulty, 2024-03-11:
Should the NTSC version of Magic Pockets be added? I seem to remember it has a different level set than the PAL version, even though it was not US-produced.

limi, 2024-10-03:
From JOTD:

> PC version has got the same levels of world 1 as the Amiga NTSC (american version, Konami).

> PC and NTSC Amiga are exactly the same version including manual protection.

> The worlds 2, 3 and 4 are identical in all versions. The only difference is in world 1.

> Level 1 is the same as the demo level (very easy) in PC/NTSC version
> In PAL Amiga version Level 1 is the Level 3 of PC/NTSC version

> Level 2, 4 and 5 are identical

> Level 3 in PAL version has no equivalent in PC/NTSC version

> Funny thing: normally you can warp to next world only from first level of each world, but with PC/NTSC version, since level 3 is the same as PAL level 1, you can warp to world 2 from level 1 AND level 3.

> The PC/NTSC version is protected by a password system, but is better for levels since you can enter a password to skip to any level, it also has slightly different controls: to make a mega-jump on a tornado/cloud you don't have to pull joystick down anymore (which is really annoying, specially in level 2 where you need the mask to avoid doing that mega-jump on a cloud, thus giving you access to some secret area)

…so it sounds like the NTSC version has better controls and a password system — so even if there is one unique level in the PAL version, we should probably make that the default version we ship — rendered in PAL60, of course — this was not designed with the NTSC PAR of 5:6 originally.

limi, 2024-10-03:
**Games that are currently rendered in 1:1 PAR NTSC instead of 5:6 PAR NTSC:**

…and also don't have the PAL version listed as the preferred one — which doesn't mean that all of these should necessarily be in 5:6 PAR NTSC, needs to be confirmed on a case-by-case basis. There are games here that should probably be PAL that fell through the cracks because they were configured as NTSC to get 60hz (e.g. Hybris) that we may have to correct.

```
game--4x4offroadracingntsc--4x4offroadracingntsc
game--moebiusntsc--moebiusntsc
game--manhunter2ntsc--manhunter2ntsc
game--mousequestntsc--mousequestntsc
game--mindwalkerntsc--mindwalkerntsc
game--zanygolfntsc--zanygolfntsc
game--links&datadisksntsc--links&datadisksntsc
game--lasersquadntsc--lasersquadntsc
game--lemmings21mbntsc--lemmings21mbntsc
game--littlecomputerpeoplentsc--littlecomputerpeoplentsc
game--nightdawnntsc--nightdawnntsc
game--nuclearwarntsc--nuclearwarntsc
game--xenon2ntsc--xenon2ntsc
game--overthenetntsc--overthenetntsc
game--ogrentsc--ogrentsc
game--outrunntsc--outrunntsc
game--oneononentsc--oneononentsc
game--kingsbountyntsc--kingsbountyntsc
game--kristalntsc--kristalntsc
game--keefthethiefntsc--keefthethiefntsc
game--killingcloudntsc--killingcloudntsc
game--khalaanntsc--khalaanntsc
game--hybrisntsc--hybrisntsc
game--hardball2ntsc--hardball2ntsc
game--hackerntsc--hackerntsc
game--harriercombatsimntsc--harriercombatsimntsc
game--superoffroadntsc--superoffroadntsc
game--spacecutterntsc--spacecutterntsc
game--spiritofexcaliburntsc--spiritofexcaliburntsc
game--spaceharrierntsc--spaceharrierntsc
game--spacewreckedntsc--spacewreckedntsc
game--shanghaintsc--shanghaintsc
game--subbattlesimulatorntsc--subbattlesimulatorntsc
game--simcityntsc--simcityntsc
game--spacequest4ntscmt32--spacequest4ntscmt32
game--stormlordntsc--stormlordntsc
game--silicondreamsntsc--silicondreamsntsc
game--speedballntsc--speedballntsc
game--skyfoxntsc--skyfoxntsc
game--stellar7ntsc--stellar7ntsc
game--silentservice2ntsc--silentservice2ntsc
game--superspaceinvadersntsc--superspaceinvadersntsc
game--spacequest4ntsc--spacequest4ntsc
game--sevencitiesofgoldntsc--sevencitiesofgoldntsc
game--streetsportsbbntsc--streetsportsbbntsc
game--spacequestntsc--spacequestntsc
game--deathbringerspotlightntsc--deathbringerspotlightntsc
game--dragonslair&escapefstntsc--dragonslair&escapefstntsc
game--dragonstrikentsc--dragonstrikentsc
game--drakkhenfilesntsc--drakkhenfilesntsc
game--dejavuntsc--dejavuntsc
game--drdoomsrevengentsc2disk--drdoomsrevengentsc2disk
game--darksidentsc--darksidentsc
game--damoclesntsc--damoclesntsc
game--darkcastlentsc--darkcastlentsc
game--downhillchallengentsc--downhillchallengentsc
game--bootntsc--bootntsc
game--ebonstarntsc--ebonstarntsc
game--elfoceanntsc--elfoceanntsc
game--rambo3ntsc--rambo3ntsc
game--redlightningntsc--redlightningntsc
game--revengeofmutantcamelsntsc--revengeofmutantcamelsntsc
game--plutosntsc--plutosntsc
game--peterbeardsleysifntsc--peterbeardsleysifntsc
game--policequest3ntscmt32--policequest3ntscmt32
[... 62 lines cut ...]
game--witwiscarmensandiegontsc--witwiscarmensandiegontsc
game--windwalkerntsc--windwalkerntsc
game--pioneerplague--pioneerplague
game--swordofaragon--swordofaragon
game--plaguentsc--plaguentsc
game--oswaldntsc--oswaldntsc
game--vortexntsc--vortexntsc
game--cosmorangerntsc--cosmorangerntsc
game--starball--starball
game--plan9fromouterspacentsc--plan9fromouterspacentsc
game--hareraisinghavocntsc--hareraisinghavocntsc
game-notwhdl--simcity2000
game--billstomatogamentsc--billstomatogamentsc
game--finalassaultntsc--finalassaultntsc
game--battleshipntsc--battleshipntsc
game--purplesaturndayntsc--purplesaturndayntsc
game--arenantsc--arenantsc
game--monkeybusiness--monkeybusiness
demo--amigadealerdemo--amigadealerdemo
game--alienfires2199adntsc--alienfires2199adntsc
game--dragonlordntsc--dragonlordntsc
game--bill&tedsexcellentadvntsc--bill&tedsexcellentadvntsc
game--turbosprint--turbosprint
game--paladinntsc--paladinntsc
```

limi, 2024-10-05:
**These 101 games should definitely be running with a 5:6 NTSC pixel aspect ratio:**

(moved the rest to separate issues (Games that are really PAL, and Atari ST ports that should maybe be 5:6 PAR but PAL)

```
demo--amigadealerdemo--amigadealerdemo
game--4x4offroadracingntsc--4x4offroadracingntsc
game--abcmondaynightfbntsc--abcmondaynightfbntsc
game--advconstructionsetntsc--advconstructionsetntsc
game--afterburnerntsc--afterburnerntsc
game--alienfires2199adntsc--alienfires2199adntsc
game--aliensyndromentsc--aliensyndromentsc
game--amazingspidermanntsc--amazingspidermanntsc
game--amigakaratentsc--amigakaratentsc
game--archon2ntsc--archon2ntsc
game--archonntsc--archonntsc
game--arcticfoxntsc--arcticfoxntsc
game--arkanoidntsc--arkanoidntsc
game--bardstalentsc--bardstalentsc
game--battleshipntsc--battleshipntsc
game--battletechntsc--battletechntsc
game--bill&tedsexcellentadvntsc--bill&tedsexcellentadvntsc
game--blueangelsntsc--blueangelsntsc
game--bootntsc--bootntsc
game--breachntsc--breachntsc
game--bubblebobblentsc--bubblebobblentsc
game--buckrogersntsc--buckrogersntsc
game--castleofdrbrainntsc--castleofdrbrainntsc
game--castleofdrbrainntscmt32--castleofdrbrainntscmt32
game--centurionntsc--centurionntsc
game--chessmaster2100ntsc--chessmaster2100ntsc
game--cosmicreliefntsc--cosmicreliefntsc
game--deathbringerspotlightntsc--deathbringerspotlightntsc
game--dejavuntsc--dejavuntsc
game--downhillchallengentsc--downhillchallengentsc
game--dragonlordntsc--dragonlordntsc
game--dragonslair&escapefstntsc--dragonslair&escapefstntsc
game--dragonstrikentsc--dragonstrikentsc
game--drdoomsrevengentsc2disk--drdoomsrevengentsc2disk
game--ebonstarntsc--ebonstarntsc
game--f19stealthfighterntsc--f19stealthfighterntsc
game--falcon&missiondisksntsc--falcon&missiondisksntsc
game--fighterduelprontsc--fighterduelprontsc
game--fightingsoccerntsc--fightingsoccerntsc
game--finalassaultntsc--finalassaultntsc
game--fourcrystalsoftrazerentsc--fourcrystalsoftrazerentsc
game--fourcrystalsoftrazerentsc--getlegendsavedisk
game--gunshipntsc--gunshipntsc
game--hackerntsc--hackerntsc
game--hardball2ntsc--hardball2ntsc
game--hareraisinghavocntsc--hareraisinghavocntsc
game--harriercombatsimntsc--harriercombatsimntsc
game--keefthethiefntsc--keefthethiefntsc
game--kingsbountyntsc--kingsbountyntsc
game--links&datadisksntsc--links&datadisksntsc
game--littlecomputerpeoplentsc--littlecomputerpeoplentsc
game--manhunter2ntsc--manhunter2ntsc
game--mindwalkerntsc--mindwalkerntsc
game--moebiusntsc--moebiusntsc
game--monkeybusiness--monkeybusiness
game--mousequestntsc--mousequestntsc
game--nuclearwarntsc--nuclearwarntsc
game--ogrentsc--ogrentsc
game--oneononentsc--oneononentsc
game--outrunntsc--outrunntsc
game--overthenetntsc--overthenetntsc
game--paladinntsc--paladinntsc
game--peterbeardsleysifntsc--peterbeardsleysifntsc
game--pioneerplague--pioneerplague
game--platoonntsc--platoonntsc
game--policequest3ntsc--policequest3ntsc
game--policequest3ntscmt32--policequest3ntscmt32
game--policequestntsc--policequestntsc
game--princeofpersiantsc--princeofpersiantsc
game--qixntsc--qixntsc
game--rambo3ntsc--rambo3ntsc
game--redlightningntsc--redlightningntsc
game--sevencitiesofgoldntsc--sevencitiesofgoldntsc
game--shanghaintsc--shanghaintsc
game--silentservice2ntsc--silentservice2ntsc
game--silicondreamsntsc--silicondreamsntsc
game--simcityntsc--simcityntsc
game--skyfoxntsc--skyfoxntsc
game--spacecutterntsc--spacecutterntsc
game--spaceharrierntsc--spaceharrierntsc
game--spacequest4ntsc--spacequest4ntsc
game--spacequest4ntscmt32--spacequest4ntscmt32
game--spacequestntsc--spacequestntsc
game--spacewreckedntsc--spacewreckedntsc
game--spiritofexcaliburntsc--spiritofexcaliburntsc
game--stellar7ntsc--stellar7ntsc
game--streetsportsbbntsc--streetsportsbbntsc
game--subbattlesimulatorntsc--subbattlesimulatorntsc
game--swordofaragon--swordofaragon
game--thunderbladentsc--thunderbladentsc
game--totaleclipsentsc--totaleclipsentsc
game--treasuresavagefrontntsc--treasuresavagefrontntsc
game--turbosprint--turbosprint
game--typhoonthompsonntsc--typhoonthompsonntsc
game--ultima4ntsc--ultima4ntsc
game--universe3ntsc--universe3ntsc
game--vortexntsc--vortexntsc
game--windwalkerntsc--windwalkerntsc
game--witwiscarmensandiegontsc--witwiscarmensandiegontsc
game--worldgamesntsc--worldgamesntsc
game--zanygolfntsc—zanygolfntsc
```

commit 8b77cdf0 (2024-10-05): Fix: 100 additional games now have the correct 5:6 NTSC pixel aspect ratio

This fixes #167.

Games updated:

* 4x4 Off-Road Racing
* ABC Monday Night Football
* Adventure Construction Set
* After Burner (Sega)
* Alien Fires 2199 A.D.
* Alien Syndrome
* Amiga Karate
* Archon: The Light and the Dark
* Archon II: Adept
* Arcticfox
* Arkanoid
* The Bard's Tale: Tales of the Unknown
* Battleship
* BattleTech: The Crescent Hawk's Inception
* Bill & Ted's Excellent Adventure
* Blue Angels: Formation Flight Simulation
* Breach
* Bubble Bobble
* Buck Rogers XXVc: Countdown to Doomsday
* Castle of Dr.  [... cut]
closed 2024-10-05 by limi