amigavision/AmigaVision #172: Indy The Last Crusade hangs after walking around a bit

rsn8887, 2024-03-22:
If I load Indy The Last Crusade, sometimes it hangs forever after walking around for a few seconds, or sometimes it hangs forever after clicking on the copy protection icons.

limi, 2024-03-26:
That does sound like it would be an upstream issue. I’ll give it a try later, but file a bug at http://mantis.whdload.de/bug_report_page.php

limi, 2024-05-29:
Filed here: http://mantis.whdload.de/view.php?id=6501

limi, 2026-10-17:
I can’t reproduce this. I tried walking around for a few minutes, interacting with people, walking outside and got treated to a section of storytelling, etc — both on MiSTer and in Amiberry — and everything seems to work fine. Could you re-test and see if it still happens?