amigavision/AmigaVision #178: Biing! - Main Game is unplayable because of screen flickering (German Game)

Turricaner, 2024-04-20:
The Intro is ok.
But in the Main Game every time you choose a option with the mouse
the screen flickers very hard.

Retro-Nerd, 2024-04-29:
The game supports DBLPAL flickerfree monitor preset in 640x512 (AGA at least). Then it works without filckering. Well that's how i start it via Workbench (with this mode set in screen prefs).

Turricaner, 2024-05-01:
Ok - solved the problem with more ram but ...
If you look at the "DamienD's WinUAE Collection" i think it is the better solution.
No "Interlace flicker" and at last it´s faster loading. (Game HD install - no WHDLoad).

limi, 2024-05-28:
For my own reference, from the WHDLoad readme:

> The installed program requires 2.0 MB ChipMem and 3.5 MB OtherMem, but more
 seriously you should use [PRELOAD](https://www.whdload.de/docs/en/opt.html#Preload) which requires 40MB of fastmem

Turricaner, 2024-05-30:
Thanks a lot.