amigavision/AmigaVision #180: Additional missing games from the forums

limi, 2024-05-02:
> This game is missing - a really nice Arkanoid clone:
> https://www.lemonamiga.com/games/details.php?id=3520
> 
> Here is a working download link:
> https://www.amigafrance.com/forums/topic/dossier-casse-brique-sur-amiga/#post-4293

> Jump‘n Bump and Neon World would be nice in the next release, too :)
> 
> https://github.com/earok/jump_n_bump_scorpion/raw/main/Builds/jump_n_bump_LATEST.adf
> https://github.com/cobour/neon-world/

> Would also be nice for the next release:
> 
> Absolute Zero:
> https://www.lemonamiga.com/games/details.php?id=4888
> 
> Ooze The Escape:
> https://h4plo.itch.io/ooze
> 
> Cyberpunks 2 Next Generation:
> https://www.lemonamiga.com/games/details.php?id=4862

> I noticed that the normal "Alien Breed" WHDLoad version is missing. This one has proper patched CD32 controls and cheats. Only the SE seem to be there, and the Story disk.

cwscws, 2024-05-12:
https://amiga.abime.net/games/view/oids

limi, 2024-08-15:
> I noticed that the normal "Alien Breed" WHDLoad version is missing. This one has proper patched CD32 controls and cheats. Only the SE seem to be there, and the Story disk.

A configuration for this was added.

cwscws, 2024-08-24:
Boulder Däsh
http://www.geit.de/eng_boulderdaesh.html

Orfax, 2024-08-31:
Arazok's Tomb (a not so great adventure game)
https://www.lemonamiga.com/games/details.php?id=81

Mean 18 (quite a nice golf game)
https://www.lemonamiga.com/games/details.php?id=3192

Both of these do not have a WHDLoad installer available, but I think both can be installed to a hard drive.

lamprey13, 2025-03-10:
Genetic Species
https://www.lemonamiga.com/games/details.php?id=1852
https://www.myabandonware.com/game/genetic-species-m53

Simon the Sorcerer II: the Lion, the Wizard and the Wardrobe
https://www.lemonamiga.com/games/details.php?id=2348

lamprey13, 2026-09-06:
Solomon's Key - https://www.lemonamiga.com/game/solomons-key
I didn't see a WHDLoad package for it, but I did find an ADF (well DMS) in this set:
https://archive.org/details/commodore-amiga-update-86.7z

limi, 2026-10-19:
Configurations for all of these have been added, except for:

- Simon the Sorcerer II — runs too slow on real Amigas
- Genetic species — over 300MB in size, too large

closed 2026-10-19 by limi