amigavision/AmigaVision #186: Add Shadow of the Beast trainers back

Striderryu, 2024-06-06:
In early revisions Shadow of the Beast had trainers like infinite energy, infinite lives, etc...  but recent revisions trainers for this game is completely gone. Its a pretty difficult game, please add them back.

limi, 2024-06-09:
We reverted to an older version since the newer version were showing bugs and artifacts, but we can certainly have a look again, and file upstream bugs if it hasn’t been addressed.

limi, 2024-06-09:
Oh, and looks like there is a built-in infinite energy toggle with `F1` in that version, since infinite energy always can get you stuck in certain areas. Doesn’t fix the CD32 button-to-jump support, but you can do that with a joystick remap in MiSTer (and Analogue Pocket too, I believe). In case you want to play it with those two capabilities right now. :)

limi, 2026-10-20:
Looks like it might be related to whether you skip the intro or not, we should re-test and update the WHDLoad container with a note not to skip the intro if it works: https://github.com/MiSTer-devel/Minimig-AGA_MiSTer/issues/101#issuecomment-4870214049

closed 2026-10-20 by limi