amigavision/AmigaVision #188: Chaos Engine AGA - CD32. version

ReadyPlayer29, 2024-06-08:
Could we have Chaos Engine AGA and Chaos engine AGA CD32 please??

Same but different but would be nice to see Chaos Engine complete

limi, 2024-06-09:
Are there any substantial reasons (different levels, different music) other than the botched AGA re-coloring? I haven’t played the AGA version super far into the game, but I assume it has the same gameplay?

ReadyPlayer29, 2024-06-09:
CD32 version without the audio on attract mode is not as good, least AGA has title music.

limi, 2024-06-10:
Yeah, we generally go for the AGA version over CD32 when available — but in the case of Chaos Engine we went with the ECS version, as the AGA version seems like a lazy over-saturated get-the-money upgrade. But if there are other redeeming versions of the AGA version over ECS, I am all ears!

ReadyPlayer29, 2024-06-10:
AGA version has amazing music for the title and there must be some other reframing features that I or no one else has currently thought of

Do it for the music

limi, 2024-07-04:
Note to self: Music in the ECS and AGA versions is identical.

limi, 2024-08-06:
Closing since there doesn’t seem to be any other substantial differences.