amigavision/AmigaVision #195: Cursor (arrow) keys doesn't work on the MegaAGS GUI and the games/apps that use cursor (arrow) keys controls.

mattx1986, 2024-07-13:
After updating to latest AmigaVision 2024.06.06 I can't use the arrow keys to control the GUI anymore. The arrows don't work in the software in the MegaAGS collection too. Maybe it's some commodity or patch in the AmigaOS interfering with the keyboard mappings. I'm using FS-UAE and arrow keys work fine in other configurations I'm running, so it's not the problem with the emulator.

limi, 2024-07-14:
Tested it on the latest FS-UAE stable build (which is our dev environment), it works fine — we would have noticed 😄 

Neither the AGS launcher or the startup-sequence have changed for many releases, so it’s likely something in your local setup?

mattx1986, 2024-07-14:
I just solved the issue by creating new FS-UAE config for MegaAGS from scratch. This is weird as the old config worked with older release just fine and the config didn't have any 'unusual' options enabled.

Thanks for the reply and excuse me for wasting your time. Maybe someone had similar issue and this post will be helpful for others though. :D

closed 2024-07-14 by mattx1986