amigavision/AmigaVision #204: Missing game in games.txt

onaryc, 2024-09-19:
I'm working on nfc card to launch amiga games through amigavision. Everything work as intended except for apidya (perhaps other are missing but i've just started to write the nfc tags) which is not in the file "games.txt" ([mister game directory]/Amiga/listings/games.txt).

I couldn't find the file in the github code so i imagine that this file is generated from the main databases.

limi, 2025-09-28:
Banshee has been reported to be missing too.

limi, 2025-09-28:
OK, the root cause is because some games are in the `Problematic` subfolder in `Run/`, and do not get picked up by the script that generated `games.txt`. Will look into a fix.

limi, 2026-10-15:
Fixed as part of #432.

closed 2026-10-15 by limi