amigavision/AmigaVision #207: Default config to NTSC

limi, 2024-10-03:
The boot time is negatively affected by the switching from PAL to NTSC (depending on the TV/monitor, this can add 1-2 seconds), so let’s try defaulting to NTSC in the MiSTer configuration. Might be a terrible idea, but let’s try it for a bit during release testing, and we’ll see if anything breaks 😄

closed 2024-10-03 by limi

don-raph, 2024-11-19:
I'm not sure if this is the right place to ask. I've found this issue because README states that the default setting for "TV Standard" should be PAL, but it was set to NTSC after a fresh install. It sounds like this is intentional, perhaps documentation is out of date?

And another question related to this topic, is this parameter supposed to be adjusted automatically when launching PAL titles? It seems to always stick to NTSC (according to core settings in MiSTer's OSD), however the TV is switching to ~50 hz refresh rate. I'm new to Amiga/Minimig, so I don't know how these video modes are handled and whether this is the correct behavior.

don-raph, 2024-11-22:
Thank you! Just to clarify - I'm using a TV via HDMI and it seems to handle 50hz correctly based on my testing, so that's what I wanted to use for Euro PAL titles. I wasn't sure if the system is indeed switching to PAL mode correctly because the OSD continues to show NTSC after launching PAL titles, like Another World:

[image]

However the TV itself is running at 50 hz, so I suppose I shouldn't pay attention to what the OSD says.

limi, 2024-11-22:
That’s a setting, not a state indicator. If you want to see what’s actually being output to the TV, open the OSD and do arrow left, and you will see both the internal resolution and the output resolution.

commit c38b42cc (2024-10-03): Default setup switched to NTSC, this closes #207