amigavision/AmigaVision #213: Some bundled apps (Protractor 3) are cropped due to NTSC mode

rsn8887, 2024-10-14:
Since the WB is loaded with NTSC Screenmode selected, some apps have GUI elements missing on the bottom, such as Protracker 3:

[image]

And Mline (although Mline scrolls automatically):

[image]

limi, 2024-10-16:
Yes, probably needs `setpal` applied, will take a look. (Workbench was always configured for NTSC, but this probably just reads the hardware default when deciding what to render)

limi, 2024-10-17:
Musicline Editor is fixed, I still need to look at PT3.

limi, 2025-05-24:
Looks like both of these do the right thing now that the system defaults to PAL in 2025.05.05

closed 2025-05-24 by limi