amigavision/AmigaVision #215: Disable Dynamic Crop option

limi, 2024-10-20:
Turns out, the 1080p `vadjust`file is applied even though you are not in a 1080p resolution. So to support other resolutions, we should offer an option to disable the Dynamic Crop 1080p/4K scaling.

There used to be a check like this:

`if NOT EXISTS ENV:dsblvadj`

And the scripts to disable/enable looked like this

```
echo "1" >ENV:dsblvadj
echo "1" >ENVARC:dsblvadj
```

and

```
delete ENV:dsblvadj >NIL:
delete ENVARC:dsblvadj >NIL:
```

They also applied a default crop, but in a way that is not compatible with the current way that is handled.

We would also need to kill the vadjust file in MiSTer: if it exists.

sjftech, 2026-08-27:
I'd definitely welcome this option as I use Direct Video to the RetroTINK-4K and want to use my own vadjust file to feed into a CRT profile and so don't want a forced 5X scale.

limi, 2026-08-28:
Direct Video shouldn’t be affected by our scaling, but I suspect that there are bugs in the Minimig core since people are reporting that it happens on CRTs too: #414

sjftech, 2026-08-28:
It definitely is affected. I've temporarily reverted to using the Minimig rbf itself without the mgl files so that it doesn't overwrite the vadjust file all the time.

limi, 2026-10-02:
Thanks for the confirmation. Let me see if I can find a fix for the Minimig core, it should definitely not apply to any resolution that is not 1080p.

closed 2026-10-15 by limi