amigavision/AmigaVision #219: Possible to add "Amiga Addict magazine Top 100" in Charts section for next release please?

simulant-systems, 2024-11-11:
Hi guys. Issue 11 of Amiga Addict magazine listed the attached Top 100 Amiga games of all time as a pull-out poster.

I hoped you may be able to add this to the AmigaVision AGS menu if possible please? It would be really cool to have a current mag that's in the shops have their chart in, along with a bit of information on the magazine alongside.

Website is https://www.addict.media

[image]

I'm not sure how these lists get made (if you do them, or the community submit them) so I'm happy to help make the files etc if needed. Apologies also for posting this in "Issues" - let me know if there is a better way to contact you.

I hope you like the idea.

Many thanks.

limi, 2024-11-12:
Yes, that certainly isn’t too hard. If you want to take a crack at it in the meantime, you can make a version of this file that lists them in this format:

https://github.com/amigavision/AmigaVision/blob/72741efee23623d590cae06e002e1f6e18d240d9/configs/Lists/Games-Charts.yaml#L212

I linked to the Amiga Power Top 100 section as it would be very similarly configured.

simulant-systems, 2026-09-08:
Just spotted a small typo: the note says "Issue 11, 2002" but it should be 2022. The magazine only started in 2021. Thanks for adding the list!