amigavision/AmigaVision #228: North & South speed issues

crusher667, 2024-12-17:
During train or fortress scenes there's a timing problem. Gameplay runs much too fast and makes game unplayable.
I tried the force PAL option but that makes no difference.
I can get the game to work in normal speed by exiting to workbench, switch video mode to PAL from right-click menu, start MegaAGS and then North&South. Game is running in normal speed - even in train or fortress scenes.
BUT when doing so the mouse cursor is moving automatically in one of the corners even if d-pad is not pressed.

limi, 2024-12-17:
Thanks for the report, will take a look!

limi, 2025-05-05:
There’s a new version of North & South in the 2025.05.05 release, let me know if this fixes the issue.

limi, 2026-10-17:
Tested on MiSTer, normal speed here.

closed 2026-10-17 by limi