amigavision/AmigaVision #230: On Amiga CD32: Can't find SYS:WHDSaves

Darkmelman, 2024-12-23:
[image]

Menu isn't starting. The Version 2024.08.08 was working fine...

limi, 2024-12-24:
Would you mind showing a screenshot of the disk configuration section in MiSTer’s OSD?

Darkmelman, 2024-12-24:
Hey!Its on the real CD32 Hardware, not Mister

Darkmelman, 2024-12-24:
I have created the folder "SYS:WHDSaves" with Opus and now, it's working!

limi, 2024-12-25:
Oh, you are running AmigaVision’s CD32 support on an actual CD32, I see 🤣

Yeah, it tries to store the save files on the Saves: HDF, which you probably do not have mounted. I’ll see if I can make it a bit smarter.

renaudguerin, 2025-01-05:
I have the same problem on a real A1200. It worked with the August release.

v00dooperson, 2025-01-15:
I'm encountering an issue with the 10.10.24 release, showing the same as the above (both on a real CD32/TF330, and a virtual A1200 under winUAE). It seems to be unable to proceed if the SYS:WHDSAVES can't be found. The 18.08.24 release shows no such issue, and proceeds as described in the AmigaVision documentation.

sh00tthec0re, 2025-03-31:
Encountered the same problem with an Amiga 4000.  Creating a SYS:WHDSaves folder got it working.

limi, 2025-04-01:
Fix upcoming.