amigavision/AmigaVision #236: Some demos need `setPAL` since the new default is now NTSC

limi, 2025-01-05:
Some of the demos that are not in WHDLoad containers get confused when the standard is NTSC. We can probably fix these with `setPAL` in the launch scripts. Currently identified as having issues:

* Coda
* Electric Lifeforms

IainMcNulty, 2025-01-10:
Running the demos in alphabetical order via [By Title], I tested "B" today.

The following exhibited issues where the demo would not run:

**Bingo:** Fails to load.

limi, 2025-01-25:
Yeah, I think there are a lot more edge cases than we expected. I think I’ll revert everything back to PAL on boot, if we are going to do this change in the future, it will require a ton of work and testing. Unlucky that I didn’t encounter any of these on my initial testing of the change.

IainMcNulty, 2025-01-25:
Question: Were all of the demos in Amigavision tested before the change was made?

I think there is still value in continuing to test so that we know which demos are incompatible. I'll be happy to continue to until you tell me otherwise.

limi, 2025-01-28:
No, I just tested a random selection of demos and games, and unfortunately they all worked, so I assumed it wouldn’t be an issue — which it turns out it is. 

I think it’s better if we just put it back to PAL by default. There’s going to be a ton of games that can run into this too, and I don’t  think it’s worth the substantial effort to get them all to work.

niallquinn71, 2025-01-29:
Apart from the fact, that most, apart from the very early USA games, it was PAL all the way, both game releases and demo releases all being done in the UK, or other Europe PAL counties. :)

limi, 2025-01-31:
No. I guess I should add this somewhere more permanent, but that is not true:

https://misterfpga.org/viewtopic.php?p=94657#p94657

rhester72, 2025-02-21:
> Apart from the fact, that most, apart from the very early USA games, it was PAL all the way, both game releases and demo releases all being done in the UK, or other Europe PAL counties. :)

That's a really common and frequent statement...but it's unfortunately also quite incorrect.

Roughly 37% of the (commercial retail) Amiga games library was in fact US publisher and developer in origin.

limi, 2025-05-05:
We have reverted to PAL as the default boot setting in 2025.05.05

closed 2025-05-05 by limi