amigavision/AmigaVision #238: Missing PD Shareware Game Beetle

PeteKeys007, 2025-01-10:
It would be great if you could add the Beetle shareware game to the collection. 

The creator of the game published it online as mentioned here: https://eab.abime.net/showthread.php?t=81396

The versions are below. I'm not sure if the WHDL file is the full version or the demo

https://ftp2.grandis.nu/turran/FTP/Non%20TOSEC%20Games/WHDLoad/Aladin%20WHDLG/Beetle-WHDL.lha
https://ftp2.grandis.nu/turran/FTP/TheZone/files/_2016/Beetle%20(incl%20extras)%20[...for%20D%20R%20Hall...].zip

PeteKeys007, 2025-01-10:
Here's a video of the game in action. https://www.youtube.com/watch?v=dWkAZhA3rgo

limi, 2025-01-11:
Seems like a decent game, will take a look at adding it

PeteKeys007, 2025-02-06:
Thanks limi

closed 2026-10-16 by limi